NVIDIA-AI-IOT/redtail #100: Output seems to be not right

mohanen, 2018-10-16:
Hi 

I ran the stereoDNN in TX2 with the image given in the sample app and the output doesn't seem to be right.

**nvidia@tegra-ubuntu:~/mohanen/stereoDNN/bin$** ./nvstereo_sample_app nvsmall 513 161 ../models/NVTiny/TensorRT/trt_weights_fp16.bin ../sample_app/data/img_left.png ../sample_app/data/img_right.png disp fp16
 

[image]
[image]

**Output**
[image]

Can anyone let me know what seems to be the problem here ..

Thanks and Regards,
Mohanen B

Alexey-Kamenev, 2018-10-18:
At the moment, the only network that correctly handles FP16 is ResNet18_2D.

The history is a little bit complicated: Stereo DNN code was originally released for TensorRT 3.0 where all networks were working fine, though not fully utilizing FP16 potential. In TRT 4.0, there were some changes which enabled proper FP16 support (and close to 2x speedup) in plugins given the plugins opt-in. We did not have time to update all of our plugins, so other networks will not work properly in FP16.

mohanen, 2018-10-23:
Ok, I am able to get the output in fp32 with NVTiny

XKXV-rose, 2018-12-19:
hi, i have a problem. when i execute nvstereo_sample_app, there is a error: cannot execute binary file.  i have do "chmod 777 file" . i cannot solve it,  could help me, thank u