NVIDIA-AI-IOT/redtail #101: Implementation of getSerializationSize and serialize

LinHungShi, 2018-10-17:
I am trying to write the engine to a plain file. It seems two related functions haven't been implemented yet. Is there any tutorial demonstrating how to implement these two functions?

LinHungShi, 2018-10-18:
Hi @Alexey-Kamenev  thanks for your reply. Since there're no weights in ELU layer, I am wondering how to serialize weight. To be more specific, do all layer variables need serialization? For example, there are dependencies between  kernel_weights and kernel_weights_d_ in Conv3DPlugin, is it necessary to serialize both of them?

Alexey-Kamenev, 2018-10-23:
In general, you should serialize all variables which are required to restore the plugin state correctly during deserialization (i.e. when deserialization ctor is called). That includes weights and biases in case of conv3d plugin. `kernel_weights_d_` is just a device copy of `kernel_weights_` so you only need to serialize one of them - `kernel_weights_`. Some variables which can be re-created do not require serialization, like various descriptors, auxiliary tensors etc.

LinHungShi, 2018-10-24:
Hi, your answer really saves my time. Do you mean I just need to serialize those important variables and recreate less important variables during deserialization?