NVIDIA-AI-IOT/redtail #102: Cannot get stream using H.265 encoder

earcz, 2018-10-18:
When I change parameters of gscam to transmit video to host, It does not stream or transmit. 
I run gscam with h265 enabled as in the guide, It gave output:
```
[ INFO] [1537034527.044330488]: Using gstreamer config from rosparam: "v4l2src device=/dev/video0 ! tee name=t ! queue ! videoconvert ! omxh265enc ! video/x-h265, stream-format=byte-stream ! h265parse ! rtph265pay config-interl=1 ! udpsink host=10.42.0.176 port=6000 t. ! queue ! video/x-raw, width=640, height=360 ! videoconvert"
[ INFO] [1537034527.056849464]: using default calibration URL
[ INFO] [1537034527.057031256]: camera calibration URL: file:///home/nvidia/.ros/camera_info/camera.yaml
[ INFO] [1537034527.060633432]: Loaded camera calibration from 
[ INFO] [1537034527.124990392]: Time offset: 1537030579.356
[ INFO] [1537034527.320534584]: Publishing stream...
[ INFO] [1537034527.321710168]: Started stream.
Framerate set to : 30 at NvxVideoEncoderSetParameterNvMMLiteOpen : Block : BlockType = 8 
===== MSENC =====
NvMMLiteBlockCreate : Block : BlockType = 8 
===== NVENC blits (mode: 1) into block linear surfaces =====
```
Thus, I looked at troubleshooting section and tried running the code:
`gst-launch-1.0 udpsrc port=6000 ! application/x-rtp, encoding-name=H265,payload=96 ! rtph265depay ! h265parse ! avdec_h265 ! xvimagesink
`
It gave a `WARNING: erroneous pipeline: no element "avdec_h265"
`
I could not solve the problem. what might be the problem ?

earcz, 2018-10-18:
I solved the problem by installing `gstreamer1.0-plugins-bad` on host. Sorry for this question

closed 2018-10-18 by earcz

vxgu86, 2019-10-02:
I could not get gstreamer1.0-plugins-bad by apt-get, said that it's replaced by gstreamer1.0-plugins-base and good now. 
how to do that?