NVIDIA-AI-IOT/redtail #105: Package not found PX4_Controller, Image_Pub, Caffe_ROS

amano001, 2018-12-04:
I followed the Wiki pretty closely, I am able to get gazbeo running with QGround Control linked in.  I did use the nvidia docker image but when I try to run the controller node, image and caffe it says package not found for any of them.  I see the folders (although they are showing archived red?)  for each of those packages inside the ws in docker.

Alexey-Kamenev, 2018-12-04:
Our Docker build script [creates](../blob/master/tools/simulation/docker/Dockerfile.kinetic#L179) symbolic links to redtail source code rather than pulling the sources directly into the container. Please check the target for those symlinks and make sure they point to the right place (i.e. redtail source code directory).

amano001, 2018-12-05:
Thanks Alexey,

I will try checking the symlinks to make sure they are pointing to the right spot.

amano001, 2018-12-06:
Hey Alexey,

Just wanted to say thanks that did the trick, I was not pointing at the correct directories.

closed 2018-12-06 by amano001