NVIDIA-AI-IOT/redtail #107: redtail docker building error about Fast-RTPS

pentaris, 2019-01-04:
Hello!

I got the following error message during the build of the Redtail Docker image.

[image]

Collecting jinja2
  Downloading https://files.pythonhosted.org/packages/7f/ff/ae64bacdfc95f27a016a7bed8e8686763ba4d277a78ca76f32659220a731/Jinja2-2.10-py2.py3-none-any.whl (126kB)
Collecting MarkupSafe>=0.23 (from jinja2)
  Downloading https://files.pythonhosted.org/packages/bc/3a/6bfd7b4b202fa33bdda8e4e3d3acc719f381fd730f9a0e7c5f34e845bd4d/MarkupSafe-1.1.0-cp27-cp27mu-manylinux1_x86_64.whl
Installing collected packages: MarkupSafe, jinja2
Successfully installed MarkupSafe-1.1.0 jinja2-2.10
You are using pip version 8.1.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0

gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now
The command '/bin/sh -c apt-get update && apt-get -y --no-install-recommends install         ant         bzip2         ca-certificates         ccache         cmake         curl         genromfs         git         gosu         iproute         iputils-ping         less         lcov         libcudnn7=7.1.4.18-1+cuda9.0         libcudnn7-dev=7.1.4.18-1+cuda9.0         libeigen3-dev         libopencv-dev         make         nano         net-tools         ninja-build         openjdk-8-jdk         patch         pkg-config         protobuf-compiler         python-argparse         python-dev         python-empy         python-numpy         python-pip         python-serial         python-software-properties         rsync         s3cmd         software-properties-common         sudo         unzip         xsltproc         wget         zip     && apt-get -y autoremove     && apt-get clean autoclean     && pip install setuptools wheel     && pip install 'matplotlib==2.2.2' --force-reinstall     && pip install cpp-coveralls     && pip install jinja2     && rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*     && cd /opt && curl http://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-5-0/eprosima_fastrtps-1-5-0-linux-tar-gz?format=raw | tar xz eProsima_FastRTPS-1.5.0-Linux/share/fastrtps/fastrtpsgen.jar eProsima_FastRTPS-1.5.0-Linux/bin/fastrtpsgen     && ln -s /opt/eProsima_FastRTPS-1.5.0-Linux/bin/fastrtpsgen /usr/local/bin/fastrtpsgen     && mkdir -p /usr/local/share/fastrtps && ln -s /opt/eProsima_FastRTPS-1.5.0-Linux/share/fastrtps/fastrtpsgen.jar /usr/local/share/fastrtps/fastrtpsgen.jar' returned a non-zero code: 2

I think there is a problem : 
```
curl http://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-5-0/eprosima_fastrtps-1-5-0-linux-tar-gz?format=raw | tar xz eProsima_FastRTPS-1.5.0-Linux/share/fastrtps/fastrtpsgen.jar eProsima_FastRTPS-1.5.0-Linux/bin/fastrtpsgen
```

Can you tell me how to fix it?

Thank you in advance.

utku537, 2019-01-07:
Same error here. Bumping for any potential solutions

SwiftGust, 2019-01-09:
I got the exact same error here too.

utku537, 2019-01-17:
`curl http://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-5-0/eprosima_fastrtps-1-5-0-linux-tar-gz?format=raw | tar xz eProsima_FastRTPS-1.5.0-Linux/share/fastrtps/fastrtpsgen.jar eProsima_FastRTPS-1.5.0-Linux/bin/fastrtpsgen`
changed http to https, now it seems to work

briannelson45, 2019-01-17:
Confirmed.  I had the same issue, and changing:

curl http://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-5-0/eprosima_fastrtps-1-5-0-linux-tar-gz?format=raw | tar xz eProsima_FastRTPS-1.5.0-Linux/share/fastrtps/fastrtpsgen.jar eProsima_FastRTPS-1.5.0-Linux/bin/fastrtpsgen

to:

curl https://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-5-0/eprosima_fastrtps-1-5-0-linux-tar-gz?format=raw | tar xz eProsima_FastRTPS-1.5.0-Linux/share/fastrtps/fastrtpsgen.jar eProsima_FastRTPS-1.5.0-Linux/bin/fastrtpsgen

in the "Dockerfile.kinetic" file (same location as build_redtail_image.sh) fixed the problem for me.  Thank you utku537!

pentaris, 2019-01-18:
@utku537 Thank you!! 

I build it except for the Fast-RTPS part.

Now. I will go back to the way you suggested.

Thanks again.

closed 2019-01-18 by pentaris