NVIDIA-AI-IOT/redtail #108: Was anyone able to recreate this Redtail project? And if so, were you successful?

daylanc, 2019-01-25:
Hello,

Just wondering, I really like this project and how it's potentially a good way to learn about NVIDIA's capabilities and UAV concepts. I was just curious as to see who has been able to successfully recreate this project. I definitely have a big interest in trying this out, but am very afraid that it may be too difficult, especially with limited community support. Has anyone been able to recreate this project, and if so, how long did it take you?

Thanks!

earcz, 2019-01-25:
I imitated the project. It took almost 5 months to achieve for me. But time depends on you. Redtail is a really challenging project which is amazing because it trains you interdisciplinary. Just for info, it would be expensive so make your research about materials.

daylanc, 2019-01-25:
@earcz 
Thanks for the input! I currently have a Titan Xp Gpu and the Jetson TX2. I'm guessing this would be the biggest cost? How much more do you think it might cost?

earcz, 2019-01-26:
Expenditure depends on the money rate in your country. For example, carrier board for jetson tx2 (I used orbitty carrier), lidar, optical flow and pixhawk 2.1 are also expensive. But not as much as jetson tx2. So you should evaluate the cost of materials. If you want to produce a drone with 3d printer, you should also consider about it because you might crash the drone a few times (I crashed 4 times). No matter what, you should keep going on this project by calculating the expenditure carefully.

daylanc, 2019-01-26:
@earcz
Thanks for the advice! Didn’t think about 3D printing a drone, but I can see how it could save some money. Did you, by chance, keep logs of your steps, progress, and issues? And if so, would you be willing to share them with me? I’d be very beneficial to see the point of view of someone who didn’t have access to expensive equipment!

earcz, 2019-01-27:
I had too many issues during the project. I documented some steps. But my documents are little bit confusing and there are a vast amount of notes about the issues. Also, you may not encounter the same problems. So your request is too general. It must be more specific. You should just start to project and solve your problems when they pops up. Because this project is realy comprehensive and deep. At first, I asked the same questions to authors in here. They told me to start from “test in the simulator”. That’s the best way to grasp the idea and concept of project. I recommend you the same thing. As for hints, I recommend you to learn installing cuda, gpu driver, digits with docker. Installing them were really challenging and time consuming since version of programs are the keypoint for building the system correctly without errors.

daylanc, 2019-01-27:
@earcz
That makes a lot of sense. A project like this will definitely require a lot of time and learning. I’ll get started with the simulation and hopefully docker will cooperate. Thank you.

earcz, 2019-01-27:
You are welcome. Good luck with the project !

mtbsteve, 2019-01-27:
> @mtbsteve
> I used Logitech c930e webcam. I also just bought a Zed camera for learning stereo logic, test it and compare it with monocular camera.

thanks. Please post your findings.

Alexey-Kamenev, 2019-01-30:
To all: first of all, thank you for your interest in the project! If you tried to reproduce it and documented your experience - feel free to post the links here, I can add those to our References section.

Second: unfortunately, we never really had time to create a proper BOM with links/prices and other detailed information though our [TBS Wiki](../wiki/Skypad-TBS-Discovery-Setup) was very close to that.
The answer [above](https://github.com/NVIDIA-AI-IOT/redtail/issues/108#issuecomment-457915039) provides a good summary worth repeating:
1. Start with the simulator on your desktop with NVIDIA GPU. No Jetson or drone required. And if you don't plan to train our models from scratch then you don't need beefy GPU, any modern GeForce would suffice.
2. Move to Jetson TX2. At this point you will need Jetson DevKit and some simple USB web camera. You can run most of the redtail components on Jetson, while still using simulator to run things like PX4/MAVROS.
3. Finally, do the drone build. When we started the project 3 years ago, we chose 3DR Iris+ as supply was abundant and it was relatively cheap and hackable. Things have changed a lot since then, but you can still find it or build your own platform from any ARF kits, if needed.

closed 2019-03-25 by daylanc