NVIDIA-AI-IOT/redtail #109: Replacement for TBS discovery drone?

daylanc, 2019-01-29:
Is there a good replacement for the TBS discovery drone? I just got off the phone with TBS and they stated that the discovery has been out of production for about 2 years now. Could someone recommend something compatible that this redtail wiki could still support? Or where they still have the discovery in stock?
Thanks

Alexey-Kamenev, 2019-01-30:
In theory you should be able to use any platform that has Pixhawk (or PixFalcon) and has enough payload capacity to lift Jetson with camera/LIDAR/PX4Flow (about 0.5kg AFAIR). Unfortunately, I cannot recommend a specific platform other than searching for relevant ARF kits (I would recommend at least 450 size). 

`redtail` itself does not really care much about the particular drone platform as long as it runs PX4 firmware and can be controlled via MAVLINK/MAVROS.

daylanc, 2019-02-01:
@Alexey-Kamenev 
Thank you for the response. I will do some research into different types of kits and see if something else will work for me.

EEGL, 2019-04-15:
@daylanc I had success with a DJI F550 kit – I couldn't buy the TBD Discovery neither, so ended up getting the F550. It still seems to be available in a few shops. I had to design and 3d print a few additional parts (to mount the Jetson TX2 and the battery) – I can share the details if you're interested.

daylanc, 2019-04-28:
@EEGL 
Yes please! That would really help me out!

briannelson45, 2019-05-22:
@EEGL Would you be willing to share some pics of how you mounted your Jetson to the F550?  Would you be willing to share designs of your 3D printed parts?

daylanc, 2019-07-12:
@eeglbalazs 
Could you share the details of your F550 build with me? Which propellers, motors, batteries, ESCs, etc. did you choose to use?

eeglbalazs, 2019-08-12:
@daylanc @briannelson45 Here are details for the F550 build [https://github.com/eegl/Redtail-F550-Build/wiki](https://github.com/eegl/Redtail-F550-Build/wiki) If you read through the TBS build in this Redtail repo and then check my instructions, that should be enough to get you going. One thing I don't remember is which screws I used to assemble the 3d printed parts.

closed 2019-10-16 by daylanc