NVIDIA-AI-IOT/redtail #111: Stuck on Simulating Controller and Joystick

daylanc, 2019-02-01:
Hello,

I am on the simulation step where we connect an Xbox controller to control the simulated drone. How exactly do I connect a wireless Xbox 360 controller or Xbox One controller to the simulation? Looking online I need a wired Xbox 360 controller or a wireless gaming receiver adapter to get it connected to ubuntu. Is there any other ways to connect a wireless Xbox 360 or Xbox One controller to the simulation?

Also, following issue #105 PX4_Controller, Image_Pub, and Caffe_Ros are highlighted red in the docker container. I was trying to follow the symbolic links, but couldn't figure out what was wrong. I was trying to edit line 179 of this code: [https://github.com/NVIDIA-AI-IOT/redtail/blob/master/tools/simulation/docker/Dockerfile.kinetic#L179](url) but didn't seem like my changes made any difference. After editing the dockerfile do we need to rebuild the redtail image? Or can we just open a new container? Maybe I'm having an issue with the pointing paths from the host directory to the container..

Thanks.

Alexey-Kamenev, 2019-02-01:
As for the controller: we use [NVIDIA Shield controller](https://www.nvidia.com/en-us/shield/shop/) which connects to a computer via Bluetooth. We also tried wired version of XBox controller and it worked fine (remember to provide [proper](../blob/master/ros/packages/px4_controller/src/px4_controller.cpp#L547) `joy_type` when running `px4_controller` node). I'm not familiar with wireless XBox controller so if it does not have USB connector then you probably need to buy adapter, as you mentioned.

As for the build: redtail Docker image does not have `redtail` source code. This is done intentionally as containers should be stateless, if possible. `redtail` source code directory is mapped into the container by providing parameters to the [run_px4_docker.sh](../blob/master/tools/simulation/run_px4_docker.sh) script, which by default are `/data/` and `/data/` which means `/data/` directory on the host is mapped into `/data/` directory in the container). Once the container is created and running, you can run `build_redtail.sh` script from the container, optionally providing path to redtail sources (default is `/data/src/redtail`). You don't need to edit anything, just make sure your paths are consistent when running the scripts.

daylanc, 2019-02-01:
@Alexey-Kamenev
Will the px_contoller node work if the controller isn’t paired correctly with the computer? I’m trying to figure out if it’s a problem with the container not having the source codes or if it’s a problem with the Xbox wireless controller not pairing properly with the computer.

daylanc, 2019-02-02:
@Alexey-Kamenev 
Solved this issue: My problem was that I wasn't removing the redtail-sim docker container correctly. I was checking for open containers using `sudo docker ps` and stopping them using `sudo docker stop redtail-sim` after checking for open containers again, it didn't show the redtail-sim container, so I assumed I closed it. The correct way to check is with: `sudo docker ps -a`
As for the symlinks problem, once the redtail-sim container was closed correctly, I could start up a new container using `sudo ./run_redtail_docker.sh redtail-sim ~/redtail/ ~/redtail/` to map the /home/redtail/ folder in the host to the /home/redtail/ folder in the container. Once inside the container, I ran `./build_redtail.sh /home/redtail/` and it successfully built all packages. checking the `/ws/src/` folder, I verified all packages were built because they turned cyan colored.
Trying the controller node worked and I can control the quad in Gazebo. For a follow-on question, what are the next steps for the simulation? Should I test the DNN and train/test my own trail?
Thank you for all the help.

Alexey-Kamenev, 2019-02-06:
Glad you were able to resolve the issues! Yes, the next steps would be trying our pre-trained model on your trail images/videos and see how it performs. I would expect pre-trained DNN to perform reasonably well if your trails are not **very** different from ours. But the very first step should be testing on a single test image to make sure all components work fine (use [image_pub](../wiki/Testing-in-Simulator#running-dnn-and-controller-nodes) and one of our [test images](../tree/master/ros/packages/caffe_ros/tests/data)).

closed 2019-03-25 by daylanc