NVIDIA-AI-IOT/redtail #112: Couldn't find executable named caffe_ros_node below /root/ws/src/caffe_ros

savath, 2019-02-03:
Hello,

I'm on the "Running DNN and controller nodes" within the "Testing in Simulator" section. I was able to run the image publisher node in step 2 but when i tried running the DNN node in step 3, i got an error saying:

[rosrun] Couldn't find executable named caffe_ros_node below /root/ws/src/caffe_ros

I went into the /root/ws/src/caffe_ros and confirmed that there was no executable file name caffe_ros_node but when i went further into /root/ws/src/caffe_ros/src, i see a file name caffe_ros_node.cpp. I was wondering if this is the file that needs to be in /root/ws/src/caffe_ros? If not, how would i fix the error?

 Please see attached image for a screen shot on this issue.

[image]

savath, 2019-02-06:
@Alexey-Kamenev 
We were not able to solve the above issue. We tried to run `catkin_make` from the catkin workspace and ran into the following build errors:

[image]

After doing some research, we think it could be a problem with OpenCV or TensorRT? We currently do not have OpenCV installed on the host and have TensorRT 4.0.1.6 as a .tar file (which was used to build the docker image). Are we missing a step in, which is the cause of our rosrun errors?

Alexey-Kamenev, 2019-02-06:
Did you provide the correct version of TensroRT when building a Docker image (as documented [here](../wiki/Testing-in-Simulator#building-docker-image))? You need to use TRT 4.x as `redtail` is not tested with later versions and may not work. In general, if you build the image using our scripts/instructions then you don't need to install anything else. If you already built an image without proper version of TRT then follow instructions in the [previous response](../issues/112#issuecomment-460905298).