NVIDIA-AI-IOT/redtail #116: Problem with connecting to Nvidia shield controller via bluetooth on Ubuntu

chao92, 2019-03-07:


chao92, 2019-03-07:
[image]
Does anyone know how to connect the Nvidia joystick to the Ubuntu Via Bluetooth?

chao92, 2019-03-07:
I tried to use Blueman, it still does not work.
[image]

Alexey-Kamenev, 2019-03-07:
We did not have any problems connecting our Shield controllers on Ubuntu 16.04.
Here is what I see now on my Ubuntu 18.04:
```
[NEW] Device 00:04:4B:77:F1:6C NVIDIA Controller v01.04
[CHG] Device 00:04:4B:77:F1:6C Connected: yes
[NVIDIA Controller v01.04]#
```
Try connecting the controller using USB cable and see if it works.

chao92, 2019-03-08:
Thank you so much for your prompt response. I just updated my system from 16.04 to 18.04. It could be able to connect right now. but it will automatically disconnect. Did you have ever meet this problem?
[image]

closed 2019-03-21 by chao92