NVIDIA-AI-IOT/redtail #117: Simulated Quadcopter: Where to find urdf and xacro file

daylanc, 2019-03-19:
@Alexey-Kamenev 
I want to modify the gazebo simulation by adding a camera node onto the simulated iris quadcopter. I understand the URDF or XACRO file needs to be modified to add the camera, but I cannot find the file in the docker container: `~/px4/Firmware/` 

Could someone point me in the right direction on where I could find the urdf or xacro file for the iris quadcopter? Currently, I am looking in the redtail-sim docker container after running `build_redtail.sh`.

Thanks.

Alexey-Kamenev, 2019-03-21:
Hello,
Those files can be found in [PX4 sitl_gazebo](https://github.com/PX4/sitl_gazebo) repo which is pulled as a submodule of [PX4 Firmware](https://github.com/PX4/Firmware/tree/master/Tools). PX4 Firmware is pulled as a part of our `redtail-sim` Docker. 
Note that we haven't had a chance to upgrade PX4 to later versions so redtail is still using very old one, 1.4.4. See [redtail-sim Docker](../blob/master/tools/simulation/docker/Dockerfile.kinetic) for more details.

daylanc, 2019-03-25:
@Alexey-Kamenev 
Because the px4 firmware is pulled after running `~/build_redtail.sh /data/src/redtail` inside the docker container, does this mean that any changes we make to px4 XACRO files will be deleted if the docker container is removed?
Also, regarding the PX4 version, how easy/hard is it to upgrade? Would it be as easy as changing the version in the [Dockerfile.kinetic](https://github.com/NVIDIA-AI-IOT/redtail/blob/master/tools/simulation/docker/Dockerfile.kinetic)? Or would it also require a lot of other changes as well?

Alexey-Kamenev, 2019-04-15:
Clarification: PX4 firmware is cloned during Docker image build (see [here](../blob/master/tools/simulation/docker/Dockerfile.kinetic#L111)) and stored in the home directory of the container so it's **not** shared among the containers and/or host. Any changes made in this directory will be lost if you delete the container. You can either change the Docker script and rebuild the image, or, which is likely simpler in your case, just clone the repo in the host directory accessible from the container and copy your files there. You can also fork PX4 repo and and commit/push your files there to make sure you don't lose any of your changes.

As for upgrading PX4 firmware: I wish it was as simple as changing the version :( Historically, the reason we decided to stick with 1.4.4 (even though the newer version was available at that time, 1.5.2 AFAIR) was stability of the firmware when flying our Irises. Newer firmware sometimes caused some strange behavior which we did not have time to debug. Also, be aware that newer versions will have different parameter names which might make the upgrade process a bit complicated.

lpmrz, 2020-12-29:
I checked [PX4 sitl_gazebo](https://github.com/PX4/sitl_gazebo) repo but it contains only SDF files. Maybe I am wrong. Are there any URDF files of Iris? Reason being, for motion planning MoveIt requires URDF files of the robots. Thanks in advance.