NVIDIA-AI-IOT/redtail #121: Running redtail on a Nvidia NANO

patrickpoirier51, 2019-04-04:
With the release of the NANO, it would be ineresting to test it with redtail.

Here are the specs: 
https://developer.nvidia.com/embedded/buy/jetson-nano-devkit
GPU | 128-core Maxwell
CPU | Quad-core ARM A57 @ 1.43 GHz
Memory | 4 GB 64-bit LPDDR4 25.6 GB/s
Storage | microSD (not included)

Considering the architecture, is this a practical project ?

kozlu, 2019-04-11:
I agree with you...

Alexey-Kamenev, 2019-04-15:
Jetson Nano is essentially Jetson TX1 with half of the GPU, that is, 128 CUDA cores in Nano instead of 256 in TX1, so our TrailNet model should run fine (but slower). Long time ago we did a performance comparison of `redtail` running on TX1 vs TX2 [here](http://on-demand.gputechconf.com/gtc/2017/presentation/s7172-nikolai-smolyanskiy-autonomous-drone-navigation-with-deep-learning.pdf) (slide 24), you can see that SResNet runs in about 11ms per frame in FP16 so even if Nano runs the network 2 times slower, this still should be enough to run the model at 30fps.

akrolic, 2019-06-09:
I am currently most of the way through implementing this project on the Nano, and have ran into difficulties. Mainly due to the fact that the Nano has to use jetpack 4.2 and this project heavily depends on jetpack 3.2/3.3 aswell as ubuntu 16.04. I am using a docker container on the Nano that is running ubuntu 16.04 and I was able to install everything so far up to the caffe_ros_node. I am currently having difficulties getting tensorRT 4.0 installed inside of my docker container.

gskachkov, 2019-08-31:
@akrolic have you managed to fix issues with docker?

akrolic, 2019-08-31:
I was not able to get past the issues with Tensor RT but it was suggested to me by other people that I could try to extract the .deb package for tensor RT from Jetpack. When you are installing jetpack onto the TX2, you first download the packages on your host machine and then push those packages onto the TX2. If you are familiar with flashing the J120 firmware, you would know where to access these .deb packages. Copying the .deb package from there is the only way I am aware of in order to have a version of Tensor RT that is installable on a arm64 machine (like the TX2 and Nano). I will post an update when I get time to work more on this project again. But it might be a while

mtbsteve, 2019-09-30:
@akrolic I got redtail adopted to and running with Jetpack 4.2.2 including TensorRT on a TX2 along with a ZED camera. 
As far as I tested no issues yet. 
I installed everything directly, no docker necessary. 
Here is the link to my git:
https://github.com/mtbsteve/redtail
Instructions are in the JP4.2 install script

gskachkov, 2019-09-30:
@mtbsteve  Great to hear this! I'll tried to play with your repo. 
Thanks for sharing this news!!!

akrolic, 2019-09-30:
@mtbsteve @gskachkov I will look at your repo as well steve, what version of ubuntu are you running? Still 16.04 or bionic?

mtbsteve, 2019-10-02:
@akrolic I am running Ubuntu 18.04 along with ROS melodic and Jetpack 4.2.2 on the TX2.

patrickpoirier51, 2019-10-02:
@akrolic  Are you still working on implementing on the NANO ?

akrolic, 2019-10-02:
@patrickpoirier51 Hey Patrick, Good to hear from you. 
I lent the Nano out to a fellow student so she can complete a project this semester. 
I might purchase another Nano to continue this development. It would be great to have this project working on the smaller platform. I will post an update here as soon as I begin working on it. 

If anyone else is interested in supporting this effort to get Redtail fully implemented and optimized for the Nano let it be known here. :D

arush, 2020-03-07:
I am interested in testing it on the Nano. I will try to go through @mtbsteve ’s TX2 instructions and see if it works

gskachkov, 2020-03-08:
@arush  Cool! I've checked it on Nano on Ubuntu 18 without docker and it works fine!

mssentop, 2021-07-06:
Hey there! Is there a way to connect Jetson Nano to PixHawk 2.4.8? I have been unable to get PixHawk working on Jetson Nano since I failed to connect it to a topside computer. Any ideas?