NVIDIA-AI-IOT/redtail #122: Using redtail DNNs to follow a path in a building, Intel T265 tracking camera

peterwatsonnm, 2019-04-09:
Hello,

I am exploring the possibility of using the resources provided by this awesome github page to design a drone to navigate autonomously within a building and locate a target.  Please correct me if I'm wrong, but this is my understanding of the project: When the drone is following a trail, its only goal is to stay on the path while moving forward and detecting obstacles.  The drone I am designing needs to be able to know which way to go, if, for example, it runs into a T in a hallway.  (it is in a gps denied environment).  Is it at all feasible to modify the DNN to be able to do this? If, for example, the path the drone had to follow had already been filmed, could I make some kind of analyzer that would interpret it for the DNN to search for?  

I have also been considering using the new Intel RealSense t265 tracking camera in conjunction with the redtail resources.  It uses VIO and SLAM systems to let the drone know its position in an occupancy map.  

I am not very experienced with DNNS but I can build robots and I am very dedicated to the project.  Thanks for your help and patience!

nlona, 2019-04-15:
Drone flying in indoor environnent? Just provide some propelleer guards. I anche inexperienced but It looks to me that deep learning AI /CV problems are with leaves (see the studio AI selflying drone). Indoor there should not be no problem.

Alexey-Kamenev, 2019-04-15:
> When the drone is following a trail, its only goal is to stay on the path while moving forward and detecting obstacles. 

Correct - the initial goal of `redtail` project was to provide only very basic autonomous capabilities like staying in the channel (e.g. trail). We did explore adding more practical features like biasing DNN with control input or high-level navigation commands (similar to turn-by-turn navigation) but did not have time to complete the project (and make it "releasable").
Some ideas were explored in other work, most notably [this](https://arxiv.org/abs/1710.02410) (for ground vehicles).

In the indoor environment with maps and VO/SLAM the usefulness of purely DNN-based navigation diminishes - you basically already have everything you need without DNNs. They can still be useful as an additional input to the controller providing redundancy and robustness in case other systems fail.

peterwatsonnm, 2019-04-15:
Thank you for your input!  I will explore the paper that you linked.  The connection between turn by turn navigation and biasing the DNN with control input was one that I had not made.  That will be useful for my project.  Thanks again!  And, thank you nlona for providing insight about flying indoors.

utkugulgec, 2019-04-29:
I want to ask a related question before the issue gets closed. What would a drone using TrailNet DNN do when it encounters crossroads on the trail?