NVIDIA-AI-IOT/redtail #123: Improvements to StereoDNN?

vivkr, 2019-04-30:
Will improvements to StereoDNN be made to improve performance on the TX 2 anytime soon?

Also, I was wondering if it would be possible to implement YOLO v3 in the StereoDNN ROS node? If so, could you please give me tips on how I could update the StereoDNN node?

Thanks!

Alexey-Kamenev, 2019-04-30:
We currently do not have any improvements planned for StereoDNN.

If you are interested in Jetson-tested object detection models, [this repository](https://github.com/NVIDIA-AI-IOT/tf_trt_models#od) should hopefully help. The repo has SSD model which, for any practical scenario, should work as well as YOLO.

vivkr, 2019-05-06:
@Alexey-Kamenev If I'm using a ZED camera, do I need to re-train the model or can I just use your models from the repo? And what resolution of images from the ZED camera gives the best depth accuracy? (Sorry for the novice questions!)