NVIDIA-AI-IOT/redtail #126: any hint whether can bu run on Jetpack 4.2?

vxgu86, 2019-07-07:
I could not even run jetson_ros_install.sh, it stuck somewhere, should it only can be used in Jetpack  3.2?

mtbsteve, 2019-08-16:
that script tries to install ros kinetic which is not supported on Ubuntu 18.04.
You may try to install ros-melodic instead.
Likely you need Ubuntu 16.4 along with JP 3.3 for Redtail.

vxgu86, 2019-09-09:
for the tx1 or tx2,they tested on :
JetPack 3.2 which contains CUDA 9.0 and TensorRT 3.0.
JetPack 3.3 which contains CUDA 9.0 and TensorRT 4.0.

for Nano, the JetPack 4.2  contains:
TensorRT 5.0.6.3
CUDA 10.166
cuDNN 7.3.1.28

these products should also go with lower version of TersorRT and CUDA, right?
I am trying on this.

mtbsteve, 2019-09-29:
I got it running with ros-melodic on Ubuntu 18.04 and Jetpack 4.2.2
Here is the link to my github:
https://github.com/mtbsteve/redtail/blob/master/ros/scripts/ardupilot_jetson_JP42_ros_install.sh
Note it’s WIP and frequently updated

vxgu86, 2019-10-07:
thanks @mtbsteve ,did you test it all with the launches in simu and real drone?

mtbsteve, 2019-10-07:
@vxgu86 I am currently only working with real drone along with the stereo_dnn_ros and strereo_dnn_ros_viz packages. I have not touched yet the simulation. 
Test cases are a mixed bag. Most of them run well with the exception of resnet18 (NVsmall and resnet18-2d run well). Also, I was not able to get gscam working. But since I use the ZED-ROS wrapper I think I will not need gscam.

vxgu86, 2019-10-07:
thanks, I will try it on my nano.
by the way, gscam not working due to ubuntu version , 18.04 not 16.04?

vxgu86, 2019-10-10:
get gscam work on 18.04, right now trying to replace caffe_ros with other node that support 18.04. 
I think will be a good start to fork redtail to 18.04

closed 2019-10-10 by vxgu86