NVIDIA-AI-IOT/redtail #129: Segmentation Fault - PC with Ubuntu 16.04, Cuda 9

ktnguyen2, 2019-07-17:
Trying to run DNN nodes on Host PC (Ubuntu 16.04, Cuda 9.0, TensorRT 4.0, GTX 1080).
Successfully built with catkin_make in workspace.
Getting segmentation fault errors when running both TrailNet and YOLO
Looked at issue #80, but it did not work for me.
```
$ rosrun caffe_ros caffe_ros_node __name:=yolo_dnn _prototxt_path:=/home/nvidianav/redtail/models/pretrained/yolo-relu.prototxt _model_path:=/home/nvidianav/redtail/models/pretrained/yolo-relu.caffemodel _input_layer:=data _output_layer:=fc25
[ INFO] [1563378947.792976151]: Starting Caffe ROS node...
[ INFO] [1563378947.816678224]: Camera: /camera/image_raw
[ INFO] [1563378947.816745760]: Proto : /home/nvidianav/redtail/models/pretrained/yolo-relu.prototxt
[ INFO] [1563378947.816781938]: Model : /home/nvidianav/redtail/models/pretrained/yolo-relu.caffemodel
[ INFO] [1563378947.816829221]: Input : data
[ INFO] [1563378947.816869030]: Output: fc25
[ INFO] [1563378947.816904300]: In Fmt: BGR
[ INFO] [1563378947.816941525]: DType : fp16
[ INFO] [1563378947.816988668]: Scale : 1.0000
[ INFO] [1563378947.817037836]: Shift : 0.00
[ INFO] [1563378947.817080789]: Cam Q : 1
[ INFO] [1563378947.817118014]: DNN Q : 1
[ INFO] [1563378947.817157823]: Post P: none
[ INFO] [1563378947.817196096]: Obj T : 0.15
[ INFO] [1563378947.817246661]: IOU T : 0.20
[ INFO] [1563378947.817286890]: Rate  : 30.0
[ INFO] [1563378947.817325582]: Debug : no
[ INFO] [1563378947.817362807]: INT8 calib src  : 
[ INFO] [1563378947.817399055]: INT8 calib cache: 
[ WARN] [1563378947.817436560]: The use_FP16 parameter is deprecated though still supported. Please use data_type instead as use_FP16 will be removed in future release.
[ INFO] [1563378948.090917109]: Loading cached model from: /home/nvidianav/redtail/models/pretrained/yolo-relu.caffemodel.cache
Segmentation fault (core dumped)
```
```
$ rosrun caffe_ros caffe_ros_node __name:=trails_dnn _prototxt_path:=/home/nvidianav/redtail/models/pretrained/TrailNet_SResNet-18.prototxt _model_path:=/home/nvidianav/redtail/models/pretrained/TrailNet_SResNet-18.caffemodel _output_layer:=out _use_fp16:=true

[ INFO] [1563378765.066485824]: Starting Caffe ROS node...
[ INFO] [1563378765.092753368]: Camera: /camera/image_raw
[ INFO] [1563378765.092836409]: Proto : /home/nvidianav/redtail/models/pretrained/TrailNet_SResNet-18.prototxt
[ INFO] [1563378765.092874193]: Model : /home/nvidianav/redtail/models/pretrained/TrailNet_SResNet-18.caffemodel
[ INFO] [1563378765.092911908]: Input : data
[ INFO] [1563378765.092946689]: Output: out
[ INFO] [1563378765.092980422]: In Fmt: BGR
[ INFO] [1563378765.093013946]: DType : fp16
[ INFO] [1563378765.093060669]: Scale : 1.0000
[ INFO] [1563378765.093097196]: Shift : 0.00
[ INFO] [1563378765.093133025]: Cam Q : 1
[ INFO] [1563378765.093167666]: DNN Q : 1
[ INFO] [1563378765.093205799]: Post P: none
[ INFO] [1563378765.093241698]: Obj T : 0.15
[ INFO] [1563378765.093277177]: IOU T : 0.20
[ INFO] [1563378765.093312307]: Rate  : 30.0
[ INFO] [1563378765.093345971]: Debug : no
[ INFO] [1563378765.093380123]: INT8 calib src  : 
[ INFO] [1563378765.093414415]: INT8 calib cache: 
[ WARN] [1563378765.093449615]: The use_FP16 parameter is deprecated though still supported. Please use data_type instead as use_FP16 will be removed in future release.
[ INFO] [1563378765.370202732]: Loading cached model from: /home/nvidianav/redtail/models/pretrained/TrailNet_SResNet-18.caffemodel.cache
Segmentation fault (core dumped)```

ktnguyen2, 2019-07-18:
For anyone who encounters this problem, delete the cached model and run TrailNet again. The segmentation fault was a result of building the cached model using a different NVIDIA driver.

closed 2019-07-18 by ktnguyen2

vxgu86, 2019-09-28:
you build the DNN  on host PC, which means not the docker or jetson tx2?