NVIDIA-AI-IOT/redtail #133: Re-Training Second Head of TrailNet

ktnguyen2, 2019-08-12:
I was able to achieve 85% accuracy with the IDSIA trails dataset, and further fine-tuned using my own data. When I try to train the second head with my own data, I get very high accuracy (~98%), but when I test a list of images from my training data, the network only predicts one class.

closed 2019-09-11 by ktnguyen2

daylanc, 2019-10-16:
@ktnguyen2 
I am running into the same issue of my model only predicting one class after training both rotation and translation heads. What did you do to solve this issue?
I'm not understanding how to "freeze" the rotational parameters after training them. It seems like training both heads are exactly the same except for the pretrained model for the translation head.

ktnguyen2, 2019-10-17:
@daylanc I found that if you get high enough accuracy (at least 85%), then test the model by deploying it in ROS and analyze the results based on test images that you send it through via imagepub node.

It is a very tedious and crude method of evaluation, but it's the best that I found without looking deeper into NVIDIA DIGITS.

soldierofhell, 2019-10-21:
@ktnguyen2, @daylanc,
Hi guys, I'm preparing Tensorflow2 + TensorRT on Jetson PoC for Ardupilot open-source. Are you interested in sharing your data? Of course just for PoC implementation, not for public sharing. We all know that Caffe is closed and to move this concept forward switching to TF is mandatory (seems like TF is now more advanced with TensorRT then PyTorch), Regards

ktnguyen2, 2019-10-21:
@soldierofhell I commend your efforts! Do you mean data as in my image datasets?

soldierofhell, 2019-10-21:
@ktnguyen2, I mean to my current knowledge there's no data available for lateral head so I'm asking if people who collected their own data similar as in the paper would kindly share it to prepare PoC or some reference models in TF2