NVIDIA-AI-IOT/redtail #139: WITH_CUDA=OFF question and No module named 'numpy' error during build_redtail_image.sh

vxgu86, 2019-09-28:
why use ** -D WITH_CUDA=OFF ** to compile opencv without cuda? we need that right?

and after opencv  git, there is an import error as follows, which did influence the installation, would that matter ?

-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.12", minimum required is "2.7") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable exact version "2.7.12") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable exact version "3.5.2") 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named 'numpy'

vxgu86, 2019-09-29:
I saw the manually installed opencv in https://github.com/NVIDIA-AI-IOT/redtail/wiki/Testing-in-Simulator#installing-opencv-with-cuda-support get  CUDA on as follows:
cmake -D CMAKE_BUILD_TYPE=RELEASE \
    -D CMAKE_INSTALL_PREFIX=/usr/local \
    -D WITH_CUDA=ON \

why the Dockerfile.kinetic left it off?

Alexey-Kamenev, 2019-09-30:
We had to turn off CUDA in our Docker due to the fact that in earlier releases of JetPack (3.2/3.3 AFAIR) OpenCV was compiled without CUDA support so in order to replicate this configuration in Docker, we turned off CUDA in the Docker as well. The discrepancy with build command in our wiki is an omission as even earlier version of JetPack (3.1) had OpenCV with CUDA so there was a bit switching between the configurations.

vxgu86, 2019-10-01:
got it,thanks

closed 2019-10-01 by vxgu86