NVIDIA-AI-IOT/redtail #144: tx2 onboard connection with px4 flushed by apm not connected even with mavros

vxgu86, 2019-10-07:
I saw from https://github.com/NVIDIA-AI-IOT/redtail/wiki/Dev-board-UART-and-Pixhawk-connection,that after wired we can test the serial with 
 roslaunch mavros apm.launch fcu_url:="/dev/ttyTHS2:921600" gcs_url:="udp://@YOUR_GCS_IP"

did this mean that we can refresh the pixhawk with ArduPilot flight stack?
but after refreshed, the px4controler just stuck and said nothing.
I flushed the px4 2.4.8 with ChibiOS-multirotor-ArduCopter V3.6.11 and Nuttx-multirotor-ArduCopter V3.6.11 ,neither of them can show up with  roslaunch mavros apm.launch

change back to px4fmu-v2_lpe.px4 then it's ok.

why it's not working?

mtbsteve, 2019-10-07:
The original Nvidia Redtail project was developed for the PX4 flight controller firmware. 
It will not work with the Arducopter stack. 
If you want to run it with Arducopter, you can try the code here:
https://github.com/ArduPilot/redtail
or try my Redtail fork from there.

vxgu86, 2019-10-08:
@mtbsteve , thanks for your reply, I went through your fork, that's great and I will keep watching.
but the problem is that, the flushed pixhawk firmware should at least respond to mavros, I have not gone so far to the px4 controler part.

my fireware is pixhawk 2.4.8
flushed with ChibiOS-multirotor-ArduCopter V3.6.11 and Nuttx-multirotor-ArduCopter V3.6.11
neither of them work for 
roslaunch mavros apm.launch fcu_url:="/dev/ttyTHS2:921600" gcs_url:="udp://@YOUR_GCS_IP"

but the px4fmu-v2_lpe.px4 works fine

mtbsteve, 2019-10-08:
As I mentioned above, the original Nvidia redtail code will not work with Arducopter out of the box. 
It requirs the px4 code installed on the pixhawk.
For Arducopter, you would need apsync along with mavlink_router installed. The mavros node will then listen to mavlink_router. Also, you would need the changes to the Redtail px4_controller code as included in the git I linked above.

mtbsteve, 2019-10-08:
So mavros runs well for me along with my Arducopter 3.6.10 setup.
[image]

vxgu86, 2019-10-09:
@mtbsteve the Arducopter code for px4_controller.cpp located
https://github.com/ArduPilot/redtail/blob/master/ros/packages/px4_controller/src/px4_controller.cpp 
could not get the apm stacked firmware armed.
for example, the code below:
apm stack did not even have offboard mode, it will stuck here, unless I changed the code.

if (fcu_state_.mode != vehicle_->getOffboardModeName() && (ros::Time::now() - last_request > ros::Duration(5.0)))
        {
            // Enable OFFBOARD mode
            if (setmode_client_.call(offb_setmode) && offb_setmode.response.mode_sent)
            {
                ROS_INFO("Offboard enabled");
            }
            last_request = ros::Time::now();
        }

mtbsteve, 2019-10-09:
Arducopter does not have a OFFBOARD mode. The equivalent flight mode is called GUIDED.
This is taken care in the Arducopter code of Redtail, see here:
https://github.com/mtbsteve/redtail/blob/4d5602934e6eaaa8b7b637807b41e73c93cbe3cc/ros/packages/px4_controller/include/px4_controller/px4_controller.h#L118

mtbsteve, 2019-10-09:
I just tested my install. The px4_controller node works well, it connects through mavros to the drone and then correctly initializes it before switching to GUIDED mode. So far, the code works as it should. See the message output below.
[image]

vxgu86, 2019-10-09:
thanks bro @mtbsteve .
I has not install the async nor the mavlink_router.
I will test the Arducopter version, thanks.

vxgu86, 2019-10-09:
by the way, any changes have you done in your forkcompared to https://github.com/ArduPilot/redtail?

closed 2019-10-09 by vxgu86

mtbsteve, 2019-10-09:
Plenty. Just check the commit log.  
Btw, if you want to run apsync with Jetpack 4.2.x, follow these instructions:
https://github.com/mtbsteve/companion/blob/next-tx2/Nvidia_JTX2_JP42/Ubuntu/1_create_base_image_tx2_JP42.txt
The official release doesn’t support Jetpack 4.2

vxgu86, 2019-10-10:
thanks, I will dig into the urls and learn about it.
but funny that I managed to get the original redtail load up with ArduCopter fireware,  and stopped with offboard mode.
I will share time learn about the arducopter throughly, thanks