NVIDIA-AI-IOT/redtail #148: Extended Redtail implementation for Arducopter

mtbsteve, 2020-02-29:
Just FYI in case someone is interested - 
I just completed my implementation of Project Redtail along with a few enhancements to the original project.
- Support of Jetpack 4.2.2/ROS Melodic 
- Support of the latest version of Arducopter
- Trailnet, StereoDNN and Darknet/YOLO support
- Support of Solex GCS app for full control of all needed ROS nodes w/o need for a PC or laptop
- RTSP video streaming and full control of ZED camera + a gimbal
and some more stuff.
A big thank you to the Nvidia team @Alexey-Kamenev for this great project idea!

https://github.com/mtbsteve/redtail/blob/master/README.md

Alexey-Kamenev, 2020-03-02:
Thanks a lot for keeping the project alive!
I've updated our README with project status and link to your project.

rbachtell, 2020-03-03:
Steve, I have been working on your branch for the last couple of days and am unable to get MAVLink communication with the Cube.  I am quite familiar with Apsync and Redtail and have had various versions operational since the beginning. I have tried with the TX2 set up on the development board as well as a Connect tech board (previous working with Apsync). I’ve exhausted my knowledge base and am curious if you have had any issues with MAVLink. The AP hot spot works as well as the web app.  Thanks for your efforts on this project, looking forward to working with it again.

mtbsteve, 2020-03-03:
@rbachtell no mavlink was always working fine for me. 
Which Jetpack version are you using? Is your wiring between the cube and the TX2 correct? Are you using my branch of Apsync? Did you follow my instructions? 
https://github.com/mtbsteve/companion/blob/master/Nvidia_JTX2_JP42/Ubuntu/1_create_base_image_tx2_JP42.txt

rbachtell, 2020-03-03:
Randy Bachtell

rbachtell, 2020-03-03:
JP 4.2.2, wiring is good, followed instructions and using your branch. I re-installed image on the development board and finally got communication. I must have an issue with the connect-tech carrier, flashed the 4.2.2 BSP but can’t get it to see the serial port. I’ll continue with the dev board and get everything working before going to the CT carrier 

Randy Bachtell

mtbsteve, 2020-03-04:
Excellent! The original Nvidia dev carrier board was working out of the box for me, but those 3rd party  boards can be a PITA. It took me a while to get all necessary ports enabled on my Auvidea board. Good luck!
Stephan

arush, 2020-03-08:
im currently getting this working on the Jetson Nano. So far the installation has worked. There are a few niggles here and there but all sorted. Have not yet tried to run the DNNs and ROS but comms and video with the FMUv5 autopilot and nano are working fine over mavlink-router. I've also introduced zerotier so the nano is actually on 4G LTE not wifi. Switched AP to client mode of course.