NVIDIA-AI-IOT/redtail #153: Takeoff Mode Initiated Mid-air when starting px4_controller node in Navigate Mode

akrolic, 2020-06-01:
We were testing our drone a second time after our first successful autonomous flight when we encountered what seems to have been a bug. Let me provide some background information about our configuration so that it is clear what our pre-autonomous flight procedure was. 

We are not using a GPS but we do have a lidarlitev3HP in order to hold our altitude. We are running PX4 1.7.0 and have configured our parameters according to the wiki. We did not begin offboard mode with the drone on the ground, rather we wanted to test the DNN while the drone was already in the air and at a fixed altitude. 
Here are the steps we followed setting up the drone to execute the DNN autopilot. 
1. We took off in Stabilized Mode and then switched to Altitude mode
2. Once we stabilized the desired altitude and the px4_controller node was running we switched into offboard mode only once we received the "Switching to Navigate" message. 

This worked well the first time, but the second time we did not notice the Takeoff mode message that was sent prior to the "switching to navigate" message. 

The drone was set into offboard mode  AFTER the "switching to navigate" message but as soon as the drone went into offboard mode it appears that it tried to takeoff while already in the air. The flight logs for both the successful flight and the crash are provided in the links down below. No one was injured and the drone only lost a leg (of the many lost already). We will perform some bench tests to replicate the problem and update this thread here. If anyone has any experience with this issue or can confirm our suspicions as to what we think the problem was that would be appreciated. 

We are considering adding a GPS to the vehicle next time we fly to be able to use Position mode when setting up the drone on the trail to execute the px4_controller node. 

[image]

Successful Flight Log -->https://logs.px4.io/plot_app?log=aadf7001-941e-4fb7-9c4f-950995b07110
Crash Flight Log--> https://logs.px4.io/plot_app?log=4b64e351-cde1-4bac-b882-d56d2c3c1ccd 

Thanks! Happy Flying :,D

Alexey-Kamenev, 2020-06-04:
I would recommend checking your command line as it seems you have a typo, as well as missing colons, according to screenshot: 
[image]

it should be `_altitude_gain`, with `n` at the end. Also, both `_altitude_gain` and `_linear_speed` params should have `:` (`_linear_speed` was not set to 3.0 and instead is 2.0 - as can be seen from the output).
However, I don't think this explains behavior you are seeing. The controller state machine switches to Takeoff mode after it's armed and stays in this mode until it reaches target altitude so in case of altitude gain == 0, it should switch promptly to Navigating mode and stay there until you turn off the Offboard mode. It should not drop back to Takeoff given zero alt gain.
The way you can debug this is by removing the propellers and just simulating the drone take-off by lifting it by hand while it's running. Check the logs and output and post if you find something interesting.

akrolic, 2020-06-26:
Hey @Alexey-Kamenev Thanks for getting back to me so quickly!

We made the changes you recommended and were also surprised to find those typos in setting the parameters. We were not able to replicate the same issue again once fixing this issue. We suspect the original issue we were having were related to instability in our GPS measurements while switching into position flight mode. USB3 interference is likely the cause for this interference and uncertainty in our GPS data and when we made the switch into this mode it would cause the drone to suddenly jerk in an attempt to hold a measured position far from its true position. 

We have remedied the GPS issue and have successfully launched the drone on the trail. We have been moving forward and are looking into @mtbsteve's redtail enhanced stereo navigation implementation. We are interested in continuing the development of this project but in order to do so with our other packages we needed to migrate to ubuntu 18.04, ROS melodic and Jetpack >= 4.2 to take advantage of the new nvidia-docker features. 

Thanks again!

closed 2020-06-26 by akrolic