NVIDIA-AI-IOT/redtail #154: Unable to complete build_redtail_image.sh script

kfuller96, 2020-07-14:
Hi There,
I've run the build_redtail_image.sh script to install the TensorRT-4.0.1.6.Ubuntu-16.04.4.x86_64-gnu.cuda-9.0.cudnn7.1.tar.gz file. The install fails when it gets to the setup tools stage. 
How do I rectify this?
I've listed the output on screen when it fails

Thanks a lot in advance :) 

*************Output log*******************

Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 41 not upgraded.
Collecting setuptools
  Downloading https://files.pythonhosted.org/packages/2f/8e/38259f4a44944a92068d5ff77230511a4c685604b47a81318f9e5cf2cc24/setuptools-49.2.0.zip (2.2MB)
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "setuptools/__init__.py", line 21, in <module>
        import setuptools.version
      File "setuptools/version.py", line 1, in <module>
        import pkg_resources
      File "pkg_resources/__init__.py", line 1380
        raise SyntaxError(e) from e
                                ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-Re_B_G/setuptools/
You are using pip version 8.1.1, however version 20.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
The command '/bin/sh -c apt-get update && apt-get -y --no-install-recommends install         ant         bzip2         ca-certificates         ccache         cmake         curl         genromfs         git         gosu         iproute         iputils-ping         less         lcov         libcudnn7=7.1.4.18-1+cuda9.0         libcudnn7-dev=7.1.4.18-1+cuda9.0         libeigen3-dev         libopencv-dev         make         nano         net-tools         ninja-build         openjdk-8-jdk         patch         pkg-config         protobuf-compiler         python-argparse         python-dev         python-empy         python-numpy         python-pip         python-serial         python-software-properties         rsync         s3cmd         software-properties-common         sudo         unzip         xsltproc         wget         zip     && apt-get -y autoremove     && apt-get clean autoclean     && pip install setuptools wheel     && pip install 'matplotlib==2.2.2' --force-reinstall     && pip install cpp-coveralls     && pip install jinja2     && rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*     && cd /opt && curl http://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-5-0/eprosima_fastrtps-1-5-0-linux-tar-gz?format=raw | tar xz eProsima_FastRTPS-1.5.0-Linux/share/fastrtps/fastrtpsgen.jar eProsima_FastRTPS-1.5.0-Linux/bin/fastrtpsgen     && ln -s /opt/eProsima_FastRTPS-1.5.0-Linux/bin/fastrtpsgen /usr/local/bin/fastrtpsgen     && mkdir -p /usr/local/share/fastrtps && ln -s /opt/eProsima_FastRTPS-1.5.0-Linux/share/fastrtps/fastrtpsgen.jar /usr/local/share/fastrtps/fastrtpsgen.jar' returned a non-zero code: 1

hildebrandt-carl, 2020-11-11:
Any updates on this? I am running into the same issue.

hildebrandt-carl, 2020-11-11:
Look at my pull request:
https://github.com/NVIDIA-AI-IOT/redtail/pull/157

It fixes the docker issues. There were a few commands that needed changing in the `Dockerfile.kinetic` file.

Alexey-Kamenev, 2020-11-17:
Thank you - I've merged the fix to master.

closed 2020-11-17 by Alexey-Kamenev