NVIDIA-AI-IOT/redtail #156: Dataset used for training

Diksha-Moolchandani, 2020-11-11:
What is the dataset used for training the NVsmall, NVTiny, and ResNet models?

Alexey-Kamenev, 2020-11-11:
For training and validation we used [KITTI dataset](http://www.cvlibs.net/datasets/kitti/raw_data.php). We used [Stereo 2015](http://www.cvlibs.net/datasets/kitti/eval_scene_flow.php?benchmark=stereo) benchmark dataset only for evaluation, i.e. we did not train/fine-tune models on that dataset (it's too small anyway, 200 samples AFAIR).

Diksha-Moolchandani, 2020-11-12:
What is the significance of input size for these models? The inputs in KITTI 2015 training set are of size (368, 1232). And the input size that is mentioned in ResNet-18_2D is 513x257. How am I suposed to run it?
./nvstereo  resnet18_2D  1232 368  stereoDNN/models/ResNet-18_2D/TensorRT/trt_weights.bin left_image_path right_image_path ./disp

or

./nvstereo  resnet18_2D  513 257  stereoDNN/models/ResNet-18_2D/TensorRT/trt_weights.bin left_image_path right_image_path ./disp

In any case the ground truth will be 1232x368 and then how will I find the error if I use the second method for inference?