NVIDIA-AI-IOT/redtail #159: Jetpack 4.4 caffe_ros node Assertion failed

mtbsteve, 2020-12-02:
I upgraded to Jetpack 4.4 and now trailnet produces the following error when I launch the caffe_ros node. 

```
apsync@apsync:~$ roslaunch caffe_ros ap_zed_ros_trailnet_yolo_robot.launch
.....
[ INFO] [1606923230.702484788]: [TensorRT] Layer(SoftMax): softmax, Tactic: 12, softmax reformatted input 0[Half(3,1,1)] -> softmax[Half(3,1,1)]
[ INFO] [1606923230.702557396]: [TensorRT] Layer(Reformat): softmax copy, Tactic: 0, softmax[Half(3,1,1)] -> out[Float(3,1,1)]
[ INFO] [1606923230.702627797]: [TensorRT] Layer(Reformat): softmax_t copy, Tactic: 0, softmax_t[Half(3,1,1)] -> out[Float(3,1,1)]
[ INFO] [1606923230.963594267]: Done building.
[ INFO] [1606923230.970012629]: Saving cached model to: /home/apsync/redtail/models/pretrained/TrailNet_SResNet-18.caffemodel.cache
[ INFO] [1606923231.384133229]: [TensorRT] Deserialize required 215901 microseconds.
[ INFO] [1606923231.391384686]: Created CUDA engine and context.
[FATAL] [1606923231.391583056]: ASSERTION FAILED
	file = /home/apsync/ws/src/caffe_ros/src/tensor_net.cpp
	line = 22
	cond = dims.type[0] == DimensionType::kCHANNEL

[trails_dnn-4] process has died [pid 12067, exit code -6, cmd /home/apsync/ws/devel/lib/caffe_ros/caffe_ros_node __name:=trails_dnn __log:=/home/apsync/.ros/log/ae48429c-34b3-11eb-baee-00044bc54178/trails_dnn-4.log].
log file: /home/apsync/.ros/log/ae48429c-34b3-11eb-baee-00044bc54178/trails_dnn-4*.log

```
Everything runs well on Jetpack 4.2. Any idea anyone what I can do here to make it tun on JP4.4?

EhrazImam, 2021-09-21:
Hi, @mtbsteve so have you gone through the stereoDnn part when i am running the inference code on jetpack 4.4 it show me the error of Opencv required version 3.3.1 and i am having by default  4.1.1..

EhrazImam, 2021-09-23:
can you help me how to install opencv 3.3.1 instead of 4.1.1 which come by default with sdk