NVIDIA-AI-IOT/redtail #160: 2.4 GHz range cut due to Jetson RF noise

gregdronsky, 2021-04-21:
Hi,
we are working on a Jetson Nano, that is mounted on a drone, and we are having problems with the 2.4GHz range. When Jetson boots up, the range of Futaba 2.4 with Fasst protocol shortens by a lot, also 2.4 Wifi card located 20cm from Jetson has problems with the range due to signal noise caused by Jetson. Did you have any experience with that? We are not connecting any Wifi card to Jetson, only Ethernet cable. Also during the boot up of Jetson, it seems that GPS signal is being disturbed as well. Can anyone help me with that? I am out of ideas about what could be the cause.
Cheers!

mtbsteve, 2021-04-21:
You need to mount the Jetson as far as possible away from the RC Receiver antennas and in particular away from the GPS. In addition, you may shield the Nano with EMI absorbing fabric.  
Example source:
https://store.mrobotics.io/product-p/3m-emishield01-mr.htm

I am running a Jetson TX2 w/o any issues on my drone after I applied the 3M foil.

gregdronsky, 2021-04-21:
Hi mtbsteve,
thank you for you fast reply. Should I place the fabric in some particular place or just cover it all up in it? It is kind of expensive on farnell, do you think it will work for sure?

nv-jeff, 2021-04-21:
We also found that USB 3 will interfere with the GPS signal as well as wifi (see here for an older investigation from Intel: https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html)  If I recall correctly, we used metal foil wrapped around the USB 3 cable to reduce EM interference.

gregdronsky, 2021-04-21:
Hi nv-jeff,
we have USB 3.0 connections. We will wrap it up with foil. Thanks!

mtbsteve, 2021-04-21:
Yes, the USB3 port is actually the main source of EMI. I have applied the 3M foil right below the GPS module. So a 6x6 centimeters  piece is fully sufficient. 
The Rx antennas are mounted under the drone  belly, with the carbon frame plates in between the Jetson and the antennas. 
I use the 5GHz wifi for the connection with my GCS, so also no interference with the 2.4RC either.

gregdronsky, 2021-04-21:
That is valuable info mtbsteve. Thank You!